ham-radio-software/D-Rats #253: Files - list (on radio) not showing

maurizioandreotti, 2023-10-07:
InThe list of files which should appear when one is connected into the files

the following is a log of what apperared on a file transfer to my station with lates Drats 4 from a station with 0.3.10
the guy was able to push a file to my pc, but was not getting the file list from mine

I wasnt able to get connected to his pc and also to get his list of files

radio connection was on VHF at less than 1km distance, so no issue on TX/RX side

10/07/2023 14:50:29:INFO:SessionManager:incoming: Received block 0:0 for session `chat'
10/07/2023 14:50:36:INFO:SessionManager:incoming:Received frame for station `!'
10/07/2023 14:50:41:INFO:SessionManager:incoming: Received block 0:0 for session `chat'
10/07/2023 14:50:58:INFO:SessionManager:incoming: Received block 0:0 for session `chat'
10/07/2023 14:51:04:INFO:StationsList:_menu_handler: executing position request to: IZ2QDO
10/07/2023 14:51:04:INFO:RPCSession:Sending job `Position Request' to IZ2QDO
10/07/2023 14:51:04:INFO:RPCSession:Job sent
10/07/2023 14:51:04:INFO:Transporter:send_frames: [SERIAL COM4@9600] Waiting 0.6 sec before transmitting
10/07/2023 14:51:05:INFO:SWFSerial:is_xon: Assuming IXANY behavior
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA10'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2Q'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>A'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>AP'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTA'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/1'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/12'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/0'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/00926'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/00926.'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/00926.31E>'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/00926.31E>B'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/00926.31E>BN  C'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/00926.31E>BN  CI'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/00926.31E>BN  CIAO*2'
10/07/2023 14:51:09:INFO:Transporter:_match_gps: [SERIAL COM4@9600] Didn't match:
b'\x11$$CRCA103,IZ2QDO>APRATS,DSTAR*:/125107h4547.62N/00926.31E>BN  CIAO*24'
[... 189 lines cut ...]
10/07/2023 14:57:09:INFO:RPCSession:Job sent
10/07/2023 14:57:10:INFO:Transporter:send_frames: [SERIAL COM4@9600] Waiting 1.1 sec before transmitting
10/07/2023 14:57:40:INFO:RPCSession:worker: Cancelling job 3 due to timeout
10/07/2023 14:57:40:INFO:RemoteFileV:_file_list_cb : Incomplete job
10/07/2023 14:59:09:INFO:SessionManager:incoming: Received block 0:0 for session `chat'
10/07/2023 15:00:48:INFO:RPCActionSet:Position request for `IZ2LXI-W'
10/07/2023 15:00:48:INFO:RPCActionSet:Position Report: Self=<rpc.RPCActionSet object at 0x0000019693673d80 (d_rats+sessions+rpc+RPCActionSet at 0x0000019693c10210)>
10/07/2023 15:00:48:INFO:RPCActionSet:rpc_pos_report: port is : COM4
10/07/2023 15:00:48:INFO:RPCActionSet:rpc_pos_report: fix is: IZ2LXI-W relazione 45.8004,9.4284@0 ft at 14:49:19
10/07/2023 15:00:48:INFO:RPCActionSet:rpc_pos_report: fix in NMEA GGA is: $GPGGA,150048,4548.026,N,925.706,E,1,3,0,0,M,0,M,,*51
IZ2LXI-W,BN  *26

10/07/2023 15:00:48:INFO:RPCActionSet:rpc_pos_report: fix in APRS is: {'rc': 'True', 'msg': '$$CRCD781,IZ2LXI-W>APRATS,DSTAR*:/130048h4548.03N/00925.71E>\r\n'}
10/07/2023 15:00:48:INFO:SessionManager:incoming: Received block 1:0 for session `rpc'
10/07/2023 15:00:48:INFO:Transporter:send_frames: [SERIAL COM4@9600] Waiting 1.3 sec before transmitting
10/07/2023 15:01:02:INFO:SessionManager:incoming: Received block 2:0 for session `rpc'
10/07/2023 15:01:02:INFO:RPCSession:__job_state: Job state: complete for 2: {'version': '0.4.1.dev39-g0fe2752-dirty', 'os': "Windows 10 (10, 0, 19045, 2, '')", 'pyver': '3.10.12', 'pygtkver': 'None', 'gtkver': '3.24.38'}
10/07/2023 15:01:02:INFO:Transporter:send_frames: [SERIAL COM4@9600] Waiting 1.3 sec before transmitting
10/07/2023 15:01:04:INFO:SWFSerial:is_xon: Assuming IXANY behavior
10/07/2023 15:01:10:INFO:SessionManager:incoming: Received block 0:0 for session `chat'
10/07/2023 15:01:50:INFO:MainWindow:_delete handler invoked <class 'gi.overrides.Gdk.Event'>
10/07/2023 15:01:50:INFO:MainApp:Saving config...
10/07/2023 15:01:50:INFO:Transporter:send_frames: [SERIAL COM4@9600] Waiting 1.8 sec before transmitting

____________

wb8tyw, 2023-10-07:
Can you reproduce this via a Ratflector?

Looks like something in the doing a 'File List Request' is not right.
To diagnose this, need to capture a "sniff" of the filelist request from a D-Rats .3 system.
I do not have the ability to locally run D-Rats .3 any more.

Two systems running D-Rats can have their serial ports connected for via a null-modem cable to allow testing with out actually needing a radio connection.

However it should also be possible to use a ratflector for testing.

maurizioandreotti, 2023-10-08:
>Can you reproduce this via a Ratflector?
In the past I had tested this and it was working.  It still is:  Connecting
from 0.4.1.dev to an host using 0.3.9 on sttammany.ratflector.com:9000 I
get the file list.

So issue happens only via RF. I did tests yesterday with 2 guys, one with
0.3.9 and one with 0.3.10beta and between them it was working correctly.

regardless of the issue we were able to TX and RX a file between 0.3 and
0.4 versions.

wb8tyw, 2023-10-08:
The receive code on through the serial port has a bug in it that appears to be in the python2 version also.

The bug is as follows:

D-Rats sends data to the radio via serial port:
D-Rats then goes to read data from the radio via the serial port.
If the radio has not yet passed the data from the remote radio to D-Rats:
  BUG: part one.  D-Rats does not check to see if it is an XOFF state!
  BUG: part two:  D-Rats does not have any code to test if the radio has requested an XOFF state.
  D-Rats assumes that the Radio has sent an XOFF to D-Rats and waits for the radio give it a single character.
  BUG: part three: If the character is not an XOFF or XON character it is discarded.

The python3 code has a small bug where it logs that this character is being discarded, which is really the only difference in this code between 0.4 and 0.3.

Note that pyserial can be configured to handle the XON/XOFF transparent to D-Rats.
This feature in pyserial has been explicitly disabled in D-Rats for unknown reasons.
We do not know why this pyserial XON/XOFF handling was disabled.  Was it ever tested?

So this code has does not work properly in either 0.3 or 0.4 versions of D-Rats.

Actions item #1:  Test if pyserial xon/xoff will work.
   Need change the places in the code where "xonxoff=0" to "xonxoff=True".
   Change the broken is_xon to always return True.
 
 Related:
 https://github.com/ham-radio-software/D-Rats/issues/134
 https://github.com/ham-radio-software/D-Rats/issues/132

closed 2023-10-21 by wb8tyw