ham-radio-software/D-Rats #255: First attempt to edit a com port on Windows threw an exception

wb8tyw, 2023-10-07:
I tried to check the configuration for a serial port, and got a exception popup instead.

On the second try, the configuration dialog came up.

Also note that the name assigned to a radio port can not be changed.

maurizioandreotti, 2023-10-11:
In Windows10, when I try adding a new port I see in the console:
```
10/11/2023 13:52:08:INFO:Win32Platform:list_serial_ports
Traceback (most recent call last):
  File "C:/msys64/home/mauri/D-Rats-master/d_rats/dplatform_win32.py", line 136, in list_serial_ports
    port = win32file.CreateFile(  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 ```

anyhow this is not blocking the creation of the com

If I try to edit an existing port, it works but in console  i see:
 ```
10/11/2023 13:56:30:INFO:DratsRadioPanel:Values: ('com3', False, 'COM3', '9600', False, False, 'com3')
10/11/2023 13:56:36:INFO:Win32Platform:list_serial_ports
Traceback (most recent call last):
  File "C:/msys64/home/mauri/D-Rats-master/d_rats/dplatform_win32.py", line 136, in list_serial_ports
    port = win32file.CreateFile(  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 ```

wb8tyw, 2023-10-11:
Unfortunately I did not record the exception in the ticket..

Along with the line of code that throws the exception we need the name of the exception that got thrown in order to fix this.

maurizioandreotti, 2023-10-11:
I have updated the previous post with more text from the console

wb8tyw, 2023-10-12:
The log entry is intentional to allow us to map the various exceptions that seem to sometimes occur.
Error code 121 is now known and will be added to the list of exceptions conditions that will not be shown in the future.