ham-radio-software/D-Rats #257: File transfer  - Mac - unxpected folder being created

maurizioandreotti, 2023-10-08:
tested on Mac, ratflector only . it could happen on other systems as well

In files tab, as soon as I connect to a remote  machine an unexpected folder is being created in the directory which hosts the source files. If this has to be it should happen into the configured folder , e.g. somewhere under /Users/maurizio/.d-rats-ev/

See here the folder created as soon as i connected to KG5CE...:
[image]

wb8tyw, 2023-10-09:
Definitely something that needs to be looked into.
The D-Rats runtime should not be updating any files in the source tree.

wb8tyw, 2023-10-10:
Only got as far as finding where it is creating the folder.   Not sure why it is creating the folder as it does not seem to use it for anything.

closed 2023-10-22 by wb8tyw