ham-radio-software/D-Rats #258: MacOs - Erron in creating config for Serial Port

maurizioandreotti, 2023-10-22:
with latest master if I try to create a config for a serial port I get:
```
Exception in thread Thread-9 (worker):
Traceback (most recent call last):
  File "/Users/maurizio/.asdf/installs/python/3.11.3/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/Users/maurizio/.asdf/installs/python/3.11.3/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/Master/d_rats/transport.py", line 423, in worker
    self.pipe.connect()
  File "/Users/maurizio/Virtualenvs/d-rats4/Master/d_rats/comm.py", line 660, in connect
    self._serial = SWFSerial(port=self.port,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/maurizio/Virtualenvs/d-rats4/Master/d_rats/comm.py", line 319, in __init__
    serial.Serial.__init__(self, **kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/serial/serialutil.py", line 244, in __init__
    self.open()
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument')
```

wb8tyw, 2023-10-22:
Only change was specifically enabling XON/XOFF protocol instead of specifically disabling it.

If that is what is throwing the Invalid argument, then that is a bug in the termios library in use by the Mac.   Can you look to see if there are any updates available?

I do not think that the Icom radio serial ports use any other handshaking XON/XOFF protocol, and if this is broken on Mac OS-X, it would require re-writing the D-Rats serial code to have a working XON/XOFF implementation, and that really is a complete re-write of the serial port.

The quick and dirty fallback is to provide an option to fall back to the broken XON/XOFF handler for Mac OS/X.

maurizioandreotti, 2023-10-26:
Serial port issue on Mac solved updating the system with:
`brew upgrade all`

closed 2023-10-26 by maurizioandreotti