ham-radio-software/D-Rats #259: Feature request dated build information

KP4AJ, 2023-10-22:
Hi John and development team:
Testing so far excellent (both repeater side and client side). Will test Ubuntu client today.  It seems D-Rats started somehow faster.  

Here a capture.  First I would recommend that build information maybe could include date of git merge or additional information that could help identify the code being used.  Right now it still shows v0.4.1.dev0 and lot have been improved past months.  Second not sure is the intended behavior or if is something at my setup;  I mean the part D-Rats shows online offline status.  KP4AJ-D Now Offline: b'Going offline  seems some typo is there.  A question I have, to test Spanish language I just need to bash run build.pot and set Spanish at preferences?

Again, highly appreciate the work being done with D-Rats!

Congrats!

Edfel, KP4AJ
[image]

wb8tyw, 2023-10-22:
The .dev.0 means that this is a tarball downloaded from some unknown commit of D-Rats or a PR of D-Rats.   That is intentional as the nature of Git means that we do not know what order a pull request will land, only update the D-Rats version on the commit after a release is tagged.

If you run from a checkout directory of a clone of the D-Rats repository it will report the last release tag value, the number of changes from the last tag, and a short version of the release tag, which will uniquely tag the version.   In some places various Python or package versioning rules may change the format slightly.

There is now an "about" or "help about" that shows the full version as known.

KP4AJ, 2023-10-22:
Thanks John! Will test git with checkout.  

Edfel

closed 2023-10-22 by KP4AJ

maurizioandreotti, 2023-10-23:
This info is interesting and need to be put in a wiki page. I do it.