ham-radio-software/D-Rats #260: Broadcast Text Message and selecting image file crash receiving station

KP4AJ, 2023-10-22:
Hi:
Well, selecting a image file (jpeg) when invoking File > Broadcast Text File cause D-Rats crash on receiving station. 

Edfel, KP4AJ

KP4AJ, 2023-10-22:
Hi:
Wonder if this could (in addition to avoid program crash) be a new feature to show images displayed at chat window.  Thinking about weather conditions, floods and other emergency - related.  Guess image size have to be minimal. 

Not sure if it is possible within D-Rats design. 

Edfel

wb8tyw, 2023-10-22:
Lots of reasons that could happen.  Would need a lot more information to troubleshoot.
D-Rats was never designed for sending large files.

* D-Rats 0.3 sends the file size in native endian which is little endian on Microsoft, and most Linux and newer Mac, and big Endian on PowerPC Macs.  D-Rats 0.4 always sends the file size in little endian.

* D-Rats may be trying to keeps multiple copies of the file in virtual memory and that could exhaust the memory in available.

* D-Rats 0.3 will randomly discard bytes from incoming packets over serial ports from radios.

* D-Rats file transfer protocol is not yet fully documented.

To troubleshoot, need a way to reproduce using the current master branch of D-Rats on both sides.

I have no way to test the 0.3 version of D-Rats anymore.

wb8tyw, 2023-10-22:
D-Rats needs a lot of internal refactoring and improvements before any significant enhancements can be done.

KP4AJ, 2023-10-22:
John, will post traceback info after returning home (have to go to the store).  Will report error happening at the sending station.   Here testbeds 1) Ubuntu Laptop 22.04 with latest D-Rats git code 2) Windows 10 with D-Rats current git code 3) Desktop with D-rats 3.10 beta 5.  

Not been able so far to reproduce first reported crash on receiving station. 

73'

Edfel
KP4AJ

KP4AJ, 2023-10-22:
John:
Was able to catch the error at the receiving station when sending a image as broadcast text from D-Rats 3.10 beta 5.  Will test what happens when the sending machine is master branch code.  

Receiving station opened a popup with the error mesage. See bellow:
Traceback (most recent call last):
  File "C:/Users/edfel/D-Rats/d_rats/mainapp.py", line 1303, in do_incoming
    self.mainwindow.tabs["chat"].display_line(line, incoming, color,

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_chat.py", line 632, in display_line
    self._display_line(line, incoming, "default", *attrs, **kwargs)

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_chat.py", line 773, in _display_line
    buffer.insert_with_tags_by_name(end, text + os.linesep, *tags)

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_chat.py", line 91, in insert_with_tags_by_name
    self.__logfile.write(text)

  File "C:/msys64/mingw64/lib/python3.11/encodings/cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

UnicodeEncodeError: 'charmap' codec can't encode characters in position 42-45: character maps to <undefined>

73'

Edfel
KP4AJ

KP4AJ, 2023-10-22:
After pressing ignore, receiving station not crashed.   So not completely sure it crashes the RX station. However, popup windows opened with the traceback message. posted above.

Edfel

wb8tyw, 2023-10-23:
An jpeg file is not a text file, so that is why attempting to send it as one is going to have problems.
Text files may only contain utf-8 characters.

wb8tyw, 2024-06-09:
This is a duplicate of https://github.com/ham-radio-software/D-Rats/issues/272

closed 2024-06-09 by wb8tyw