ham-radio-software/D-Rats #261: File Sending Slow

KP4AJ, 2023-10-22:
Hi John and Developers:
When connected via Internet (both repeater and client) I notice sending files is very slow.  Test file was just 143kb (had to close the software).  File sending should take advantage of the connection.  Not sure if it is related with latest changes.  

Setup: D-Rats client 3.10 beta 5, repeater using latest code.  Could be wrong but file sending was faster previous builds, I think. 

73'

Edfel
KP4AJ

wb8tyw, 2023-10-22:
The d-rats client is coded for half duplex communications, even for the internet connections.

There has been no change in this from the 0.3 to present master.
There probably not be any updates to the 0.3 branch for the Python 2 version.

Operating system and library speed, and internet speed may vary, and transferring files of over a few kb over a radio link will take a considerable amount of time.

KP4AJ, 2023-10-22:
Thanks for the clarification John.  Closing. 

73'
Edfel
KP4AJ

closed 2023-10-22 by KP4AJ