ham-radio-software/D-Rats #262: Chat TX/RX - DRats 0.3.10 to 0.4.1.dev18-g4406b3d-dirty

maurizioandreotti, 2023-10-22:
Today some 2hour TR/RX tests - reporting here some outcomes and console logs to support troubleshooting.

All the stations below were in same locations for both the tests scenarios with the configuration usually adopted for D-rats 0.3.x 
As all three stations were able to receive clear audio between them and all radios were showing callnames at RX time at all times the expectation was that each station was able to receive all chats when other stations were transmitting.
(i.e. everybody able to hear anybody at anytime).

Setup: 

- station FOW: with ID5100 + Win10
- station BDZ: with ID51 +Win10
- station LXI: with IC2820 + Win10 (Mac wasnt usable as unable to properly config the serial port (see issue id TBD))

Scenario A - setup and results
- station FOW: 0.3.10        -- can trasmit and receive D-Rats chat from both BDZ and FOW
- station BDZ: 0.4.1dev18 --can trasmit and D-Rats chat from both BDZ and LXI
- station LXI: 0.3.10          --  can trasmit and D-Rats chat from both BDZ and FOW

Scenario B - setup and results
station FOW: 0.3.10 with ID5100  --can receive D-Rats chat from both BDZ and FOW
station BDZ: 0.4.1dev18 with ID51  -- can receive D-Rats chat from both LXI and FOW
station LXI: 0.4.1dev18 with IC2820 -- can *not* receive D-Rats chat from other (but can transmit) (reverting to 0.3.10 the chat was receivable)

______________
Log from BDZ:

```
Administrator@FLYRAD1 MINGW64 ~/D-Rats
# ./d-rats.py
10/22/2023 16:08:06:INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
10/22/2023 16:08:06:INFO:D-Rats:main: args.config = C:/Users/Administrator/AppData/Roaming/D-RATS-EV
10/22/2023 16:08:06:INFO:Mainapp:Enabled debug log for Win32 systems
10/22/2023 16:08:06:INFO:QST:FeedParser not available
10/22/2023 16:08:07:INFO:DratsConfig:File C:/Users/Administrator/AppData/Roaming/D-RATS-EV/d-rats.co
nfig
10/22/2023 16:08:07:INFO:MainApp:_refresh_lang: Loading locale `it'
10/22/2023 16:08:07:INFO:MainApp:_refresh_lang: Locale dir is: D:/msys64/home/Administrator/D-Rats/l
ocale
10/22/2023 16:08:07:ERROR:MainApp:_refresh_lang: Messages catalog file missing  for it.  Need to use
 'msgfmt tool to generate.
10/22/2023 16:08:07:INFO:MainApp:D-RATS v0.4.1.dev17-ga0b0c90 starting at Sun Oct 22 16:08:07 2023 o
n Platform Win32Platform:
  configuration:       C:/Users/Administrator/AppData/Roaming/D-RATS-EV
  system_data: D:/msys64/home/Administrator/D-Rats
  OS version: Windows 10 (10, 0, 19045, 2, '')
10/22/2023 16:08:07:INFO:MainApp:load position from config file
10/22/2023 16:08:07:INFO:MainApp:_static_gps: Configuring the Static position: 45.820300,9.379220
10/22/2023 16:08:07:INFO:MainApp:_static_gps: Mapserver not active: False, call: IU2BDZ
10/22/2023 16:08:07:INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8203,9.3792@420 ft
at 16:08:07
10/22/2023 16:08:07:INFO:D-Rats:main: reloading app

10/22/2023 16:08:07:INFO:MainApp:load the main window
10/22/2023 16:08:07:INFO:MainApp:load main window with self config
10/22/2023 16:08:09:INFO:DRatsPluginServer:Started serve_forever() thread
10/22/2023 16:08:09:INFO:MessageRouter:_route_via_station: Starting message router thread
10/22/2023 16:08:09:INFO:MainApp:connect main window
10/22/2023 16:08:09:INFO:MainApp:connect tabs
10/22/2023 16:08:09:INFO:MainApp:going online
10/22/2023 16:08:09:INFO:MessageRouter:_run_one: Station list was empty
10/22/2023 16:08:09:INFO:MainApp:create map window object-----
10/22/2023 16:08:09:INFO:MapWindow:init MapWindow
10/22/2023 16:08:09:INFO:MainApp:create map window object: connect object-----
10/22/2023 16:08:09:INFO:MainApp:query local gps device to see our current position
10/22/2023 16:08:09:INFO:APRSicons:get_icon: Unknown APRS symbol table: x
10/22/2023 16:08:09:INFO:MainApp:invoke config refresh
10/22/2023 16:08:09:INFO:MainApp:Refreshing config...
10/22/2023 16:08:09:INFO:gps:set_units: Set GPS units to Metric
10/22/2023 16:08:09:INFO:MainApp:refresh comms
10/22/2023 16:08:09:INFO:MainApp:_refresh_comms: Re-Starting ports_0
10/22/2023 16:08:09:INFO:MainApp:Starting Comms
[... 178 lines cut ...]
10/22/2023 16:35:22:INFO:Utils.log_exception:----------------
10/22/2023 16:35:22:INFO:Transporter:__recv: [SERIAL COM6@9600] Data path IO error: Failed to read f
rom serial port [SERIAL COM6@9600] ClearCommError failed (PermissionError(13, 'The device does not r
ecognize the command.', None, 22))
10/22/2023 16:35:27:INFO:Transporter:__recv: [SERIAL COM6@9600] Attempting reconnect...
10/22/2023 16:35:27:INFO:Utils.log_exception:-- Exception: --
Traceback (most recent call last):
  File "C:/msys64/home/mauri/D-Rats-Master/d_rats/comm.py", line 717, in read
    data = self._serial.read(size)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/home/mauri/D-Rats-Master/d_rats/comm.py", line 462, in read
    return serial.Serial.read(self, size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/site-packages/serial/serialwin32.py", line 275, in read
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not r
ecognize the command.', None, 22))
10/22/2023 16:35:27:INFO:Utils.log_exception:----------------
10/22/2023 16:35:27:INFO:Transporter:__recv: [SERIAL COM6@9600] Data path IO error: Failed to read f
rom serial port [SERIAL COM6@9600] ClearCommError failed (PermissionError(13, 'The device does not r
ecognize the command.', None, 22))
10/22/2023 16:35:31:INFO:MainWindow:_delete handler invoked <class 'gi.overrides.Gdk.Event'>
10/22/2023 16:35:31:INFO:MainApp:Saving config...

```

wb8tyw, 2023-10-22:
This looks like another case where the driver XON/XOFF handling may not be working 100%.

The 0.3 version XON/XOFF is guaranteed to drop characters  There may be something we do not know about some Radios that does not work with the driver XON/XOFF handling.

Ugolupo, 2023-10-23:
iz2fow version corrections: my  d-rats 0.3.9    Python 2.7.9   GTK 2.24.10   PyGTK 2.24.0  LibXML using 0.4 KB

wb8tyw, 2023-10-27:
Just noticed from this bug report, that D-Rats was operating in Linux XANY mode, which requires that the device sending XOFF to send additional XOFFs codes if needed.

If the radio is not sending the additional XOFF codes, then this can result in data loss on transmission.

The PR-268 reverts to the previous behavior, except that incoming characters that show up instead of XON/XOFF are no longer discarded.

closed 2023-11-07 by wb8tyw