ham-radio-software/D-Rats #263: Preferences > radio IC-9700 (serial) - stays TX

KP4AJ, 2023-10-22:
Hi John and dev team:
Testing latest code, when trying Radio (serial), the Icom 9700 goes into TX and stays.  Previously builds it went TX (think sending CQ) but only momentarily. Had to turn radio off.  Not sure if latest D-Rats serial changes requires change in the IC-9700 setup.  Closing D-Rats not stopping the TX (radio have to be turned off).  Not able to do RF tests yet.  

Will do some more testing.  Was working previous builds (I mean radio setup and TX) unless and I omitting something.  

73'

Edfel
KP4AJ

KP4AJ, 2023-10-22:
Hi:
Closing this as setting other COM port and radio mode to DV, makes radio behavior normalize.  

Thanks.

Edfel

closed 2023-10-22 by KP4AJ

maurizioandreotti, 2023-10-23:
Hi Edfel,

coukd you describe what are the correct setting you used for ic9700? I do
not have that radio, but that information could be useful information to
put in a wikipage for other users benefit.

Thanks anyhow for the testing you are doing!

KP4AJ, 2023-10-24:
maurizio:
Glad to share.  IC 9700 will work with the same settings for FT8.  Users just have to be sure radio mode is set to DV and COM port if NOT the same used by the IC9700 programming software,  IC9700 driver creates two COM ports. 

Will share the IC9700 settings in another post.  Hope to post tomorrow.  I have not yet test RF connection to other D-Rats station.  But radio do TX when I type at the chat and do send.  So RF testing need to be done.  

73'

Edfel
KP4AJ

maurizioandreotti, 2023-10-24:
Thanks Edfel, 

I am adding the info here: https://github.com/ham-radio-software/D-Rats/wiki/030-Connecting-D%E2%80%90Rats-with-a-D%E2%80%90Star-radio

KP4AJ, 2023-10-25:
Hi maurizio:
I got the info about IC9700 digital setup some time ago.  Here two screenshots.  Under presets I have normal and FT8. Both seems to work.  As I have had no opportunity (yet) to test D-Rats communication with other D-Rats. I can confirm radio do the TX when I place text and click on send.  Also, I can confirm those settings work with other digital mode (FT8) even the one labeled 'normal'.  As matter of fact I happen to not be using the FT8 preset lately even for digital as normal works fine.

Hope this helps someone and maybe more testing settings could be improved.

73'

Edfel
KP4AJ
[image]
[image]

KP4AJ, 2023-10-25:
Update:  baud rate seems to accept all values (9600, 19200, 38400, 115200).  Radio sends with any of those values (D-Rats preferences > Radio.   The question is at which speed radio will send the text via RF, if indeed D-Rats allows for use of the IC9700 hardware capabilities.  Not sure about that.  So experimentation and testing could be valuable. 

73'

Edfel
KP4AJ

KP4AJ, 2023-10-25:
Another comment with the normal settings have been able to QSO FT8 mode (2M) and VarAC (VARA FM) also 2M as well as packet (soundmodem) and WinLink. 

73'

Edfel
KP4AJ

Note:
 In case of VarAC and WinLink IC9700 radio settings use FM instead of DV.

maurizioandreotti, 2023-10-25:
Thanks!
I add these to the wiki page about the radio configurations here:
https://github.com/ham-radio-software/D-Rats/wiki/030-Connecting-D%E2%80%90Rats-with-a-D%E2%80%90Star-radio