ham-radio-software/D-Rats #264: Error when sending broadcast text message but choosing image file

KP4AJ, 2023-10-22:
Hi:
This when trying to broadcast text file but sending image (jpeg). Windows 10  D-Rats latest git code.   Obviously if user choose correctly this problem should not occur.  Here the traceback:
UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 65: character maps to <undefined>

Win10 Git Code Broadcast image file:
Traceback (most recent call last):

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_chat.py", line 876, in _bcast_file
    data = file_handle.read()
           ^^^^^^^^^^^^^^^^^^

  File "C:/msys64/mingw64/lib/python3.11/encodings/cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 65: character maps to <undefined>

73'
Edfel
KP4AJ

wb8tyw, 2024-06-09:
I guess that D-Rats should catch the exception and just log (hopefully only once) that the file can not be sent as it has non-text characters.   However as with the other ticket, there is no way to know if a text file is displayable on another platform.

wb8tyw, 2024-06-09:
This is a duplicate of https://github.com/ham-radio-software/D-Rats/issues/272

closed 2024-06-09 by wb8tyw