ham-radio-software/D-Rats #265: Error sending image file to specific station

KP4AJ, 2023-10-22:
When trying to send image file to a station lsted at Stations column I get this error:
Traceback (most recent call last):

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_stations.py", line 269, in popup_menu_handler
    self._sendfile_handler(station, port)

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_stations.py", line 429, in _sendfile_handler
    fname = image.send_image(fname)
            ^^^^^^^^^^^^^^^^^^^^^^^

  File "C:/Users/edfel/D-Rats/d_rats/image.py", line 175, in send_image
    dialog = build_image_dialog(filename, img, dialog_parent)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "C:/Users/edfel/D-Rats/d_rats/image.py", line 122, in build_image_dialog
    quality = Gtk.HScale(Gtk.Adjustment.new(50, 1, 100, 10, 10, 0))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

TypeError: GObject.__init__() takes exactly 0 arguments (1 given)

Same eror happens with Win 10, Ubuntu (latest code) when sending image. TXT file was received properly from Ubuntu to WIndows (both git latest code). 

Edfel

KP4AJ, 2023-10-23:
Here an update.  The image file gets sent from 3.10 beta 5.  A popup is invoked. I tired sending the same image from Win 10 (git latets code).  Same error as before. In summary 3.10 beta 5 > latest code image file sent and received ok.  latest code Win / Ubuntu > Win / Ubuntu error shown above.  TXT files are sent/received ok latest code Win / Ubuntu.

[image]
[image]

The file is a jpeg 143k size. 

Edfel
KP4AJ

wb8tyw, 2023-10-23:
I was not aware that there was any way to send an image to a chat.  As far as I know the code only actually support utf-8 characters.
On python2, native strings are 8 bit codes.
On python3, native strings are UTF-8.

Is this a send file issue?

KP4AJ, 2023-10-23:
Yes send file issue. Guess only way to send to specific station is send file.

KP4AJ, 2023-10-23:
Some old chat software indeed had the capability to show image files at chat. Different technologies, no longer available. One example 123FlashChat.   Maybe some day it could be possible with Python.    

Regards John,

Edfel
KP4AJ

wb8tyw, 2023-10-26:
Still tracing down the code.

Your original crash though only occurs if the Pillow library is installed, and it that will need to be addressed.

The unit test which use to pass is failing for me. 

This patch will fix it locally until a PR can be created.

```text
diff --git a/d_rats/image.py b/d_rats/image.py
index 82a7f0a..883cfdd 100755
--- a/d_rats/image.py
+++ b/d_rats/image.py
@@ -118,8 +118,10 @@ def build_image_dialog(filename, image, dialog_parent=None):
     dialog.size = miscwidgets.make_choice(SIZES, False, SIZES[1])
     dialog.size.connect("changed", lambda x: update())
     dialog.add_field(_("Resize to"), dialog.size)
-
-    quality = Gtk.HScale(Gtk.Adjustment.new(50, 1, 100, 10, 10, 0))
+    adjustment = Gtk.Adjustment.new(50, 1, 100, 10, 10, 0)
+    quality = Gtk.Scale.new(orientation=Gtk.Orientation.HORIZONTAL,
+                            adjustment=adjustment)
+    # quality = Gtk.HScale(Gtk.Adjustment.new(50, 1, 100, 10, 10, 0))
     quality.connect("format-value",
                     lambda s, v: "%i" % v)
     quality.connect("change-value", set_quality, dialog)
```

The message popup about no support or resizing images is  because you do not have the python pillow module installed, that is an optional module.

Found that the STATION panel send_file or the file transfer code switches to this mode if you pass a filename ending in '.jpeg', '.jpg', '.png', or '.gif'.

If you do have pillow installed, it then is supposed to pop up a dialog to convert the image file.

It looks like if Pillow is present, it allows converting the image to a low resolution icon format for a compact format and a much smaller file.   The 0.3 windows pre-built version probably has the predecessor of the python pillow module built in.

The resulting file is sent as a normal file to the location the user has configured to collect files.  I would have to look up the defaults for that.  So that is a normal file, and does not show up in the chat window.

The over the radio D-Star serial channel for 432 Mhz and below is designed for sending short messages of 1Kb or less, not anything with 10 Kb or more, which would take quite a bit of radio transmissions to send each chunk.  So sending a large file is pretty much a denial of service attack for over radio connections.

And many ratflectors are setup to relay over radio links.

We probably need a setting to limit the sending of large files automatically.

wb8tyw, 2023-10-26:
What happened is that a minor update for GTK changed changed the Gtk.Hscale call to require using a ".new" method when passing a parameter or it will fail.   The GtkHscale class has also been deprecated and may disapear in the future.

Pretty much every place that a Gtk class is invoked with a parameter with out also using a method is subject to breaking in the future, so this is an ongoing task to look up every GTK call and verify if it is deprecated.

KP4AJ, 2023-10-26:
Thanks John, appreciate a lot your dedication and time to explain your findings.  Agree probably makes sense to control large files so RF communication not vulnerable to kind of unintended DoS.   

Edfel
KP4AJ

closed 2023-10-28 by wb8tyw

KP4AJ, 2023-10-31:
Just to tell test done and image resize dialog opens, and image get resized as well.  Seems excellent resize btw. Test done dev code Win 10 and Ubuntu 22.04.

Edfel
KP4AJ