ham-radio-software/D-Rats #272: Error broadcasting this text file (Windows latest D-Rats code)

KP4AJ, 2023-11-08:
Hi:
The error affects only Windows platform (latest code).  Not a big deal pressing ignore is enough and the file get published but as error popup window shows, I am reporting it.  I found this by accident when doing a random test. Maybe happens  because one of the symbols at the text file. Situation as follows:

Ubuntu send ok receive ok
Windows send error receive error (in receive however the file gets published at chat window)
Windows 3.1 beta 5 send ok receive ok

Windows 10 (latest D-Rats code)
Error when broadcasting:
Traceback (most recent call last):

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_chat.py", line 876, in _bcast_file
    data = file_handle.read()
           ^^^^^^^^^^^^^^^^^^

  File "C:/msys64/mingw64/lib/python3.11/encodings/cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

UnicodeDecodeError: 'charmap' codec can't decode byte 0x8d in position 1299: character maps to <undefined>

Error when receiving same file (sent from Ubuntu) both machines latest D -Rats code:
Traceback (most recent call last):

  File "C:/Users/edfel/D-Rats/d_rats/mainapp.py", line 1303, in do_incoming
    self.mainwindow.tabs["chat"].display_line(line, incoming, color,

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_chat.py", line 632, in display_line
    self._display_line(line, incoming, "default", *attrs, **kwargs)

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_chat.py", line 773, in _display_line
    buffer.insert_with_tags_by_name(end, text + os.linesep, *tags)

  File "C:/Users/edfel/D-Rats/d_rats/ui/main_chat.py", line 91, in insert_with_tags_by_name
    self.__logfile.write(text)

  File "C:/msys64/mingw64/lib/python3.11/encodings/cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

UnicodeEncodeError: 'charmap' codec can't encode character '\u03a9' in position 525: character maps to <undefined>

[T-Branch.txt](https://github.com/ham-radio-software/D-Rats/files/13291616/T-Branch.txt)

Edfel
KP4AJ

wb8tyw, 2023-11-08:
What tool created this file?

D-Rats / Python 3 is assuming that all text files are Unicode UTF-8 encoded.  By default unless special options are added, python will through and exception when bytes that are illegal in UTF-8.   We have tried to put the special options in place, but may have missed some in the conversion.

Before the adoption of UTF-8 for eight bit text use, there were alternate encodings that were used, Microsoft created a set of code pages for each locale, as the example in the trace "cp1252" and various other platforms use their own encodings, such as decmcs8 and iso*.   These encodings can have 8 bit codes inside them that conflict with Unicode.

As neither Linux or Microsoft Windows track the actual encoding of text files, there is no way to auto-detect what conversion may be needed.

Python2 by default used code pages on Windows and one of the other platform specific settings.

Non UTF-8 text files with 8 bit characters in it will not display/render correctly on systems that use different non-unicode encoding.

KP4AJ, 2023-11-08:
John:
Thanks for the info.. Not remember if I downloaded the file from somewhere but could have been a text pasted into NotePad++.  

73'
Edfel
KP4AJ

wb8tyw, 2024-06-09:
There pretty much is no way to fix this issue.

There is no way to reliably know the encoding of a text file, especially on the Windows platform.

If the text file is anything other than ASCII-7 or UTF-8, it may not show up correctly when rendered.