ham-radio-software/D-Rats #274: MobaXterm install of D-rats investigation

wb8tyw, 2023-11-08:
The MobaXterm free edition has a clause that it can be uses on computers used for commercial use, so a version can be installed on any Windows version it supports.

This product includes a Debian like linux environment that can be customized.

This ticket is for investigating if a script can be used to install a working D-Rats client.

wb8tyw, 2023-11-10:
The MobaXterm defaults to opening a terminal to its own Cygwin installation and uses a package manager that emulates apt to install packages.

It looks like getting MobaXterm to work will be similar to getting a Cygwin install to work.

While MobaXterm uses Debian apt install, Cygwin uses RedHat convention for package names.

Cygwin is missing many of the pre-built packages that D-Rats depends on, some some will need to be fetched from PyPi via pip.  These packages will be installed into /usr/local/pypi and a PYTHONPATH will be needed to access them.

maurizioandreotti, 2023-11-14:
can we close this?

closed 2023-11-15 by wb8tyw