ham-radio-software/D-Rats #276: Please verify > get version -- machine identity seems published (MobaXterm setup)

KP4AJ, 2023-11-12:
Included three captures of the get version functionality. First capture from cygwin (new MobaXterm install methodology).  Second from Get version old D-rats 3.10 beta 5 (included only for comparison purposes)  the third from Ubuntu. 

[image]
[image]
[image]

Version information should only provide platform without machine identity, IMO.  When I ran the get version from Ubuntu to Win 10 MSYS32 MINGW64 install I get 'No version response from X'.  

73'

Edfel
KP4AJ

KP4AJ, 2023-11-12:
Additional info: MobaXterm setup used was the installer version (not the portable). 

Edfel
KP4AJ

wb8tyw, 2023-11-12:
"Cygwin" is a different platform than MobaXterm embedded Cygwin as MobaXterm made significant changes in their implementation that affects scripts and program execution.

There are at least to version reports in D-Rats, an RPC get version and a ping reply.

The version report is part of the RPC protocols that we do not have full documentation for.  For sending the report, a python dictionary is created and then passed through the RPC code for transmit.  Something in that code serializes the data, and somewhere at this time, I do not know where receives that serialized data and converts it to a readable string.

Some clients are over some paths are not returning a version report, and that is something that I have not traced down yet.

It is very possible that a something is broken in the serialization and deserialization code.  Until we capture byte level traces of good and bad transfers, it may be hard to fix.

Unfortunately while the "sniff" and "raw" text settings should provide this data, in practice, they do not always provide all that is needed in a usable form.

KP4AJ, 2023-11-12:
Thanks John.

wb8tyw, 2024-06-09:
Since this ticket was originally filed, A bug was discovered in the serial port code of D-Rats were it would on occasion discard the first character of a packet rendering it unusable.
This should not affect connections over a ratflector.
That bug was fixed in current D-rats 0.4, but is not going to be fixed in the 0.3 and earlier D-Rats.

So if any one reproduces this, it is important to know if it the radio was being accessed via an on-air link.