ham-radio-software/D-Rats #280: list of stations in the map on Windows MobaXTerm

maurizioandreotti, 2023-11-13:
So, not sure how, at some point the list of stations got populated, but it doesnt get updated, not even closing and reopening d-rrats or mobaxterm.
[image]

I am running 2 D-rats 0.4x in parallel connected to the same ratflector : the one running into mobaxterm doesnt refresh the list of stations after having received their position 

I pushed the position request from the mingw d-rats as the moba doesnt send the request (see other bug opened)
i am sure that the D-rats in mobaxterm also received the positions, as I can see them into the console, 

[image]

but the same do not get displayed into the "event log" window (and yes sniffing and raw text options are enabled:
[image]

maurizioandreotti, 2023-11-14:
redone tests... it seems stations were not showing as d-rats was actually disconnected from the ratflector

INFO:SocketDataPath:connection to pldares.ratflector.com 9000
INFO:SocketDataPath:getline: Host does not require authentication
INFO:RPCSession:worker: Cancelling job 3 due to timeout

wb8tyw, 2023-11-14:
I am finding on this new Windows 11 PC, with its current power settings is going into such a deep sleep that the repeater gets disconnected, or it is because something with one of the ISPs on one end or another dropping idle sessions.

Still need to investigate if TCP/IP keepalives would help here.

Attempting a communication on a disconnected repeater results in nothing happening with the client GUI which makes it look like nothing is happening.
I think we have a ticket about the silent disconnects with out reconnections.

maurizioandreotti, 2023-11-14:
I always got disconnected after a while with all versions running on
windows. And I think on the MacOS also. I always wondered how was possible
to find stations on a ratflrctor being present and unattended, I guess
since days.

I recall there was a thread years ago in the group.io but I didn't really
had time to investigate. At that time I was mainly using a drats onto a
raspberry pi and I think wasn't suffering of this issue.

By the way to my knowledge there is only one thing to force a reconnect
without exiting d-rats (so to not loose list of stations and chats in the
chat tab), which is to go into the Prefs, edit anything so fo force a
reload and reconnect.

KP4AJ, 2023-11-15:
It is working but MobaXterm seems slower than other platforms.  I left the Map window open and most tiles and station got published. 

Maybe a note in the wiki could help.  But thats my experience,  desirable others test the MobaXterm functionality (particularly maps).  Other than that MobaXterm is a quick and easy mechanism for Windows users. 

73'

Edfel
KP4AJ

wb8tyw, 2024-04-22:
MobaXterm has Cygwin based environment.   Cygwin has an extra DLL that translates some posix APIs to Microsoft Windows APIs.
Msys2 uses a more minimal interface to native Wind32 APIs.   I do not know if that makes a difference in speed.   Someone would have to design some tests to see if there is a speed difference between one or the other.