ham-radio-software/D-Rats #288: file -> preferences -> gps -> lookup : coloums in the address results are fixed

maurizioandreotti, 2024-06-30:
During a test done with mobaxterm for PR287 i noted that the window with the lookup addresses results has fixed coloums: this doesnt allow to see the correct address to pick up the right coordinates.

in my case, loocking for mtyy qth i getseveral options:

[image]

but without enlarging the first coloumn i cannot check which one is the correct

wb8tyw, 2024-07-13:
The proper fix for this is going to require making changes to the drats/listwidget.py to have a way to tell it how wide to make each column, and that looks like a more complex job.

The behaviour can be improved by changing the order of the fields so that the address is last, so that it will automatically expand by making the box wider, instead of having to make the box wider and then expand the address field.

I am also able to get the box to expand when the search address starts getting longer than the default entry size, and the box will retain that size when you move on to the locations found section, it stays expanded.

wb8tyw, 2024-07-13:
With this change, if you select a location, and then click next for the confirm address.   The dialog box will be expanded to show it.

If you then click on back, button because you selected the wrong one, the box will remain expanded, and you should be able to see enough of the addresses to make a better selection.

wb8tyw, 2024-07-13:
Also I found that if no addresses were found, the routine crashed with a stackdump.

maurizioandreotti, 2024-07-15:
Tested on mac and it seems ok.

wb8tyw, 2024-07-28:
I do not fully understand the address format in the E-mail you sent me.

To me the "Via <name>" is something for postal routing, but may not mean anything for a GPS lookup.
So I entered in just the "##, ##### C<name> LC (Italy)" for the address.

This comes up with an address very close but not exactly to what you sent me for your Google Maps lookup of your address.

I used the set point on map function with address lookup, and noticed that it placed the looked up address in the "Name" field where logically I think it should be in the comment field.   Something : to look at later.

I have pushed a fix to stop the trace back when an address is not found.   It blanks out part of the dialog, but you can click on the "Back" button to make corrections.

maurizioandreotti, 2024-07-29:
> To me the "Via " is something for postal routing, but may not mean
> So I entered in just the "##, ##### C LC (Italy)" for the address.

"via" it the italian word which means "street"

wb8tyw, 2024-08-10:
Since we are just passing the address to the lookup service, we have no control over if it understands the address or can resolve it.

I think we have worked around the problem for now.

closed 2024-08-10 by wb8tyw