ham-radio-software/D-Rats #289: Upgraded MobaXterm 24.2 getting error running the installer

KP4AJ, 2024-07-07:
Hi John:
Upgraded MobaXterm to 24.2 and tried the d-rats installer.  The error is this:
bash d-rats_in_mobaxterm_install.sh
d-rats_in_mobaxterm_install.sh: line 7: syntax error near unexpected token `newline'
d-rats_in_mobaxterm_install.sh: line 7: `<!DOCTYPE html>'

Not sure if I forgot something or if the upgrade breaks the installer.

Thanks in advance.

Edfel

KP4AJ, 2024-07-07:
Please note D-Rats runs., just wanted to get the latest changes. 

Edfel

wb8tyw, 2024-07-07:
Edited description so that all characters are visible.

The error message indicates something went wrong with the command to download the script and instead of the script you got an HTML document as indicated from the error message instead of the script.

It turns out that the URL in the README.md file in the d-rats repository is wrong and that will need to be corrected in a future push.

Please confirm that you downloaded the URL in the [Installation via MobaXterm](https://github.com/ham-radio-software/D-Rats/wiki/010.020-Installation-of-D%E2%80%90Rats-on-Microsoft-Windows-with-MobaXterml)

KP4AJ, 2024-07-08:
Hi John:

Followed from here:
https://github.com/ham-radio-software/D-Rats/wiki/010.020-Installation-of-D%E2%80%90Rats-on-Microsoft-Windows-with-MobaXterm

No error shown now.  D-Rats updated properly. Version shown now:  0.4.2.dev2-g07b1172-dirty

Thanks John!

Edfel
KP4AJ

closed 2024-07-14 by wb8tyw