ham-radio-software/D-Rats #294: Instructions for Linux install

raymon68, 2024-09-16:
Hello.  I have been trying to install D-Rats onto a debian machine.  When I try to run pip3 on the tarball I get: <SEE BELOW>

There is no setup_version.py.  Also there is no PKG-INFO file in the d_rats sub directory. 

Is there a Linux install procedure described somewhere?  Or is there documentation on how to create a pip3-installable tar file?

Thanks,

Ray

> Processing ./D-Rats-0.4.2.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [21 lines of output]
      The setup.py is only used in building a pip installable tarball.
      The setup.py is not used for actually installing d-ratsand will not work.
      Traceback (most recent call last):
        File "/home/rmoral/.miniconda3/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/rmoral/.miniconda3/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/rmoral/.miniconda3/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-t4wmf4ag/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 332, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-t4wmf4ag/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 302, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-t4wmf4ag/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 318, in run_setup
          exec(code, locals())
        File "<string>", line 75, in <module>
        File "<string>", line 56, in default_build
      FileNotFoundError: [Errno 2] No such file or directory: 'd_rats/setup_version.py'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

wb8tyw, 2024-09-17:
A tarball generated from github is useless for pip installs.  The [Readme](https://github.com/ham-radio-software/D-Rats) file should explain how to create a pip installable tarball in more detail.

I was placing PIP installable packages in [D-Rats mailing list file area](https://groups.io/g/d-rats/) file download area but gave up on that as the primary users, Windows users, found that it took too many steps for them to safely install D-rats on windows.  So almost no one was using them.

So now the quick procedure is to just do a git clone of master branch of the repository.
And then for Debian use apt-get to install all the required packages that are listed in the README above.
Then create a symbolic link to make it easy to run the package.

I took the time to get the windows procedure to simply install MobaXterm, start a local terminal window wget a script, chmod it to be executable, and then run it for the Windows users to run D-Rats under MobaXterm from a git clone.
Pretty simple and less work for me

I just have not had time to do the same for the various Linux distributions other than to document the package lists for some of the Debian distributions in the [Readme](https://github.com/ham-radio-software/D-Rats) 

Then upgrades to D-rats are usually just using "git pull" commands.

raymon68, 2024-09-17:
Thank you for your quick response.  I understand what I have to do.Ramón J. Moral Ph.D.Sent from my iPad