ham-radio-software/D-Rats #296: smtpd discontinued vs aiosmtpd

maurizioandreotti, 2024-11-26:
trying to execute d-rats on macos with python 3.13, smtpd module seems not available  and moving to aiosmtpd requires a code review (for my understanding)

File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats/d_rats/mailsrv.py", line 29, in <module>
    import smtpd
ModuleNotFoundError: No module named 'smtpd'
.venvmaurizio@Mao-MacbookAir D-Rats %

wb8tyw, 2024-11-28:
Lots of things will need to be tested for this, and the install procedures, readme files will need an update.

geek4096, 2025-01-23:
Hey guys. Any chance there's a fix in the mix for this?  Thanks much!
73
Keith/KI5US

wb8tyw, 2025-01-24:
Currently I do not have any way to test changes to the mail modules in D-Rats.

Can you see if you can find an older version of Python to install and run as one work around.

The first pass at a fix would be this to try the old module and then change the code that references to the methods from the smtp modules check the SMTP_MODULE value and return the appropriate failure status.

That would at least stop programs from crashing until a proper fix is made.

After looking at this some more, apparently the python standard library "smtplib" is the replacement, but of course it has a new API
to write to and debug.   The use of "aiosmtp" also would require a rewrite of the code and has the disadvantage in that it requires an additional package to be installed.

SMTP_MODULE = False
try:
    import smtpd
    SMTP_MODULE = True
except: ModuleNotFoundError:
    print("No SMTP python module available."
`