ham-radio-software/D-Rats #298: [AGWPE] D-Rats should use AX.25 UI Frames

dscp46, 2025-03-12:
Hi,

D-Rats' AGW module makes an inappropriate use of AX.25 I frames (which one MAY use after having properly established an AX.25 connection-mode data link).

The issue is located there: 

https://github.com/ham-radio-software/D-Rats/blob/54bddfd4567c414e1284663567ffa3ebc8bb9fa4/d_rats/agw.py#L695-L700

To comply, we would need set the control field value to `0x13` (UI Frame, poll bit set), instead of  `0x3E` (I Frame, poll bit set) at line 699. 

This should not introduce any regression, since the AGW and AX.25 headers are stripped without any sanity check, at line 106: 

https://github.com/ham-radio-software/D-Rats/blob/54bddfd4567c414e1284663567ffa3ebc8bb9fa4/d_rats/agw.py#L103-L108

Best regards,

wb8tyw, 2025-03-17:
Issue has turned out to be more complex and affects the comm.py module also.

The problem is that doing bit manipulation on python3 UTF-8 string types as AGS and comm corrupts the UTF-8 strings.

This operation was valid in the original python2 code, for python3 only a byte type can be used.

wb8tyw, 2025-03-17:
Similar bugs may be present in aprs_dprs.py, gps.py, yencode.py, and wl2k.py as they are pushing numbers into utf-8 strings.

wb8tyw, 2025-03-17:
yencode.py is clear of issues, code is only present for in the python2 compatibility mode.

wb8tyw, 2025-03-20:
gps.py is clear of this issue.

w2lk unfortunately has this, which was legal with python2, but it risks corrupting a UTF-8 string.

`        header_str = self.__name + "\x00" + chr(len(data) & 0xFF) + "\x00"
        header = header_str.encode('utf-8', 'replace')
`

closed 2025-04-13 by wb8tyw