ham-radio-software/D-Rats #301: AntiX - error on serial connection to IC2820

maurizioandreotti, 2025-03-14:
As I am recovering the old PC of the club installing AntiX, I am installing 0.4.1.dev0
it seems to connect to the   i am getting this error

....
INFO:MapWindow:init MapWindow
INFO:MessageRouter:_run_one: Station list was empty
INFO:MainApp:create map window object: connect object-----
INFO:MainApp:query local gps device to see our current position
INFO:MainApp:invoke config refresh
INFO:MainApp:Refreshing config...
INFO:gps:set_units: Set GPS units to Metric
INFO:MainApp:refresh comms
INFO:MainApp:_refresh_comms: Re-Starting ports_0
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_1
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_2
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_3
INFO:MainApp:Starting Comms
INFO:MainApp:start_comms: Starting port ports_3 (ttyUSB0)
INFO:SWFSerial:Software XON/XOFF control initialized
INFO:MainApp:check_comms_status: Ports expected to be already started:
INFO:MainApp:check_comms_status: ttyUSB0
INFO:MainApp:check_comms_status: Checking all Ports from config:
INFO:MainApp:check_comms_status: portid ports_0
INFO:MainApp:check_comms_status: Port Local_Ratflector not started
INFO:MainApp:check_comms_status: portid ports_1
INFO:MainApp:check_comms_status: Port RAT not started
INFO:MainApp:check_comms_status: portid ports_2
INFO:MainApp:check_comms_status: Port sttammany.ratflector.com not started
INFO:MainApp:check_comms_status: portid ports_3
INFO:MainApp:check_comms_status: Port ttyUSB0 already started!
INFO:MainApp:_refresh_gps : GPS: False on @4800
INFO:MainApp:_static_gps: Configuring the Static position: 45.837910,9.404950
INFO:MainApp:_static_gps: Mapserver not active: False, call: IQ2LC
INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8379,9.4049@0 m at 22:53:02
INFO:MainApp:_refresh_map: reconfigure Mapwindow with new map
INFO:MainApp:_refresh_map: reconfigured mapurl to: https://tile.openstreetmap.de/
INFO:MainApp:_refresh_map: reconfigured mapkey to: None
INFO:MainApp:_refresh_lang: Setting language to: it
INFO:MainApp:Python pycountry package needed for selecting languages.
Exception in thread Thread-3 (worker):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/iq2lc/Scaricati/D-Rats-master/d_rats/transport.py", line 438, in worker
    self.get_input()
  File "/home/iq2lc/Scaricati/D-Rats-master/d_rats/transport.py", line 229, in get_input
    chunk = self.__recv()
            ^^^^^^^^^^^^^
  File "/home/iq2lc/Scaricati/D-Rats-master/d_rats/transport.py", line 211, in __recv
    return self.pipe.read_all_waiting()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/iq2lc/Scaricati/D-Rats-master/d_rats/comm.py", line 783, in read_all_waiting
    data = self.read(1)
           ^^^^^^^^^^^^
  File "/home/iq2lc/Scaricati/D-Rats-master/d_rats/comm.py", line 762, in read
    if self._serial.dsr:
       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 730, in dsr
    s = fcntl.ioctl(self.fd, TIOCMGET, TIOCM_zero_str)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

wb8tyw, 2025-03-15:
That crash looks to be in the distro or PyPi supplied serial port driver code.
This report is missing the most critical information from the crash dump, and that is the Python exception that was thrown.

Standard practice is do do a DSR check to validate if a device can be detected.
If we know the exception we can put a try / except block to catch and report the broken driver to the user instead of crashing D-Rats.

What type of serial port hardware is it?

Can you try a different USB serial port device?

I tried changing the serial configuration GUI to allow some control of enabling these serial port checks, but the glade editor would not allow me to make the changes, which is one of the reasons I decided to start over with drats2.

maurizioandreotti, 2025-03-17:
I copy pasted the error messages at the bottom of the console, maybe i skipped something.

it is an old pc which was used for drats on windows. (maybe xp... not sure as something happened on the harddisk which got ruined, so the only way was reinstall it with a clean,... and the only linux i was able to boot, and reinstall it, was antix) 

the prolific driver port was listed in the devices by the operating system as ttyUSB0, so I assumed the driver was working. 

the serial cable it is the original one from icom for IC-2820, it should be the OPC-478 USB 
(prolific usb- serial, mini jack 2.5mm- ic2820)

I can try next friday with my home made cable (https://iz2lxi.jimdofree.com/diy-data-cable-for-ic-2820-id-51-id-5100/)

wb8tyw, 2025-03-17:
The exception name may only show up in the popup message about the exception from the global handler that D-Rats has.
 
Alternately there are examples of try/except blocks that will call the logger with "exec_info=true".

You can put a try/except and log block around the code that crashes, to get the actual exception, then then change the except block to catch the specific exception that and just log the event, possibly at the debug level if it happens a lot.
As what this is indicating is that a part of the driver that D-Rats does not depend on is broken.
On the re-write, the configure port section will use that information to notify the end user of a broken driver.

It may be a bug in the specific version of pyserial being used, once you get the real exception, that can be searched for.

https://www.decisivetactics.com/support/view?article=db-9-connector-pinout

For DYI cables, and it commercial cables I strongly recommend:

- If the RTS/CTS signals are not used by the cable on both end, on the 9-pin connector those pins should be connected together.

- If the DSR/DTR signals are not used by the cable on both end, on the 9-pin connector those pins should be connected together.

This is the reasoning:

Software is suppose to assert the RTS signal before sending any data.  D-Rats asserts the RTS signal.

Software is supposed to assert the DTR signal when it is ready to use the serial port.   D-Rats asserts the DTR signal.

Some software requires the CTS signal to be present or it won't use the cable.   That should be a configurable option, but I have seen this hardcoded on some PC programs.  With the pins connected, that works.

Software is supposed to ignore data being received if the DSR signal is not asserted unless the software has been told to ignore the modem control signal.   D-Rats monitors the DSR signal to see if it is present, but currently does not enforce ignoring data.
With those pins connected, the software can detect if the cable is connected. on the computer side, but not the radio side.

I was not able to modify the configuration UI to allow properly configuring a serial port using the glade tool.

maurizioandreotti, 2025-03-23:
I took home the antix PC and tested D-Rats with my diy cable and ID51 radio (used 2 different cables, one with direct serial and 2 with serial to usb converter with pl2303 chip) and with this  setup it doesn't produce the error, so will need to collect the ICOM cable and retest with that one...

maurizioandreotti, 2025-03-23:
Connecting the D- rats to serial port on pc the radio goes in TX but on the other side my friend with drats doesnt  get anything
If i try to do "ping all" the radio doesnt go in tx.

 ./d-rats.py 
INFO:DratsConfig:File /home/iq2lc/.d-rats-ev/d-rats.config
INFO:QST:FeedParser not available
INFO:Spelling:Demand-opening aspell...
INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
INFO:D-Rats:main: args.config = /home/iq2lc/.d-rats-ev
INFO:MainApp:_refresh_lang: Setting language to: it
INFO:MainApp:Python pycountry package needed for selecting languages.
INFO:MainApp:D-RATS v0.4.1.dev0 starting at Sun Mar 23 18:58:34 2025 on Platform UnixPlatform:
  configuration:       /home/iq2lc/.d-rats-ev
  system_data: /home/iq2lc/Scaricati/D-Rats-master/D-Rats-master
  OS version: Linux iq2lclinux 6.1.42-antix.1-amd64-smp #1 SMP PREEMPT_DYNAMIC Mon Jul 31 18:30:05 EEST 2023 x86_64
INFO:MainApp:load position from config file
INFO:MainApp:_static_gps: Configuring the Static position: 45.837910,9.404950
INFO:MainApp:_static_gps: Mapserver not active: False, call: IQ2LC
INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8379,9.4049@0 ft at 18:58:34
INFO:D-Rats:main: reloading app

INFO:MainApp:_refresh_lang: Setting language to: it
INFO:MainApp:Python pycountry package needed for selecting languages.
INFO:MainApp:load the main window
INFO:MainApp:load main window with self config

(d-rats.py:12687): Gtk-WARNING **: 18:58:35.173: Attempting to add a widget with type GtkBox to a GtkMenuItem, but as a GtkBin subclass a GtkMenuItem can only contain one widget at a time; it already contains a widget of type GtkAccelLabel
INFO:DRatsPluginServer:Started serve_forever() thread
INFO:MessageRouter:_route_via_station: Starting message router thread
INFO:MainApp:connect main window
INFO:MainApp:connect tabs
INFO:MainApp:going online
INFO:MainApp:create map window object-----
INFO:MessageRouter:_run_one: Station list was empty
INFO:MapWindow:init MapWindow
INFO:MainApp:create map window object: connect object-----
INFO:MainApp:query local gps device to see our current position
INFO:MainApp:invoke config refresh
INFO:MainApp:Refreshing config...
INFO:gps:set_units: Set GPS units to Metric
INFO:MainApp:refresh comms
INFO:MainApp:_refresh_comms: Re-Starting ports_0
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_1
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_2
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_3
INFO:MainApp:Starting Comms
INFO:MainApp:start_comms: Starting port ports_3 (ttyS0)
INFO:MainApp:_refresh_comms: Re-Starting ports_4
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_5
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_6
INFO:MainApp:Starting Comms
INFO:MainApp:check_comms_status: Ports expected to be already started:
INFO:MainApp:check_comms_status: ttyS0
INFO:MainApp:check_comms_status: Checking all Ports from config:
INFO:MainApp:check_comms_status: portid ports_0
INFO:MainApp:check_comms_status: Port Local_Ratflector not started
INFO:MainApp:check_comms_status: portid ports_1
INFO:MainApp:check_comms_status: Port RAT not started
INFO:MainApp:check_comms_status: portid ports_2
INFO:MainApp:check_comms_status: Port sttammany.ratflector.com not started
INFO:MainApp:check_comms_status: portid ports_3
INFO:MainApp:check_comms_status: Port ttyS0 already started!
INFO:MainApp:check_comms_status: portid ports_4
INFO:MainApp:check_comms_status: Port ttyUSB0 not started
INFO:MainApp:check_comms_status: portid ports_5
[... 101 lines cut ...]
INFO:SWFSerial:is_xon: Assuming IXANY behavior
INFO:SWFSerial:is_xon: Aiee! Read a non-XOFF char: `b']'`
INFO:SWFSerial:is_xon: Assuming IXANY behavior
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'OB]"=@=@\x01=@\x16\xb2=@\x06iQ2LC~~~CQCQCQ~~ciao 2[EOB]CQ~~ci`o 3[EOB]'
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b':0b=H'
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'\x11'
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'\x11\x11\x11\x11\x11\x11\x11'
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'\x11'
INFO:StationsList:_menu_handler: executing ping all
INFO:StationsList:_menu_handler: Doing CQCQCQ ping on port ttyS0
INFO:Transporter:send_frames: [SERIAL /dev/ttyS0@9600] Waiting 1.5 sec before transmitting
INFO:StationsList:_menu_handler: executing ping all
INFO:StationsList:_menu_handler: Doing CQCQCQ ping on port ttyS0
INFO:Transporter:send_frames: [SERIAL /dev/ttyS0@9600] Waiting 1.7 sec before transmitting
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'\x11'
INFO:Transporter:send_frames: [SERIAL /dev/ttyS0@9600] Waiting 0.8 sec before transmitting
INFO:SessionManager:incoming: Received looped frame
INFO:Transporter:send_frames: [SERIAL /dev/ttyS0@9600] Waiting 1.0 sec before transmitting
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'[SOB]"=@=@\x01=@\x93\x1d=@\x07IQ2LC~~~CQCQCQ~~cciao 2[EOB_'
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'\x11\x11\x11'
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'BECT JMV\xe0,\x01P\x81Mi\x10r&'
INFO:MainApp:Showing Map Window
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'\x11'
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'*DArE'
INFO:Transporter:worker: [SERIAL /dev/ttyS0@9600] ### Unconverted data: b'\x11'

wb8tyw, 2025-03-23:
Serial cable wiring is not likely to cause a crash inside of the Pyserial module that reported.  Glad you are making progress.

maurizioandreotti, 2025-03-29:
Closed as now the Antix and Radio communicate

setting to verify on the IC-2820 :

```
DV-SET MODE    >  DV  DATA TX    >   AUTO
DV-GPS               >  GPS TX                 >    DISABLED
DV-GPS               >  GPS AUTO TX      >    OFF
```

closed 2025-03-29 by maurizioandreotti