ham-radio-software/D-Rats #306: Received Station via Radio not shown in station list

maurizioandreotti, 2025-03-25:
As part of the test the Antix PC i am trying to get gps positions messages from an ID51 to a 2820 connected to D-Rats on antix PC.

What happens is that:
- doing a "ping all" from d-rats the 2820 goes in tx, and on the ID51 I see the callname set on 2820, but the ID51 doesn't do the autoreply to the call (so I suspect that the string transmitted from the radio is not the one expected)

- trasmitting from ID51 (only voice, bno gps fix yet set ont he id51) the 2820 receives but the CALL is not shown in the list of received stations.
see below the copy of what i see in the terminal-- i would expect at least that the station name (IZ2LXI or IQ2LC4) appear on the list of the station in D-rats

```

 ./d-rats.py 

(d-rats.py:32605): dbind-WARNING **: 09:43:38.931: Couldn't connect to accessibility bus: Failed to connect to socket /run/user/0/at-spi/bus_0.0: Permesso negato
INFO:DratsConfig:File /home/iq2lc/.d-rats-ev/d-rats.config
INFO:QST:FeedParser not available
INFO:Spelling:Demand-opening aspell...
INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
INFO:D-Rats:main: args.config = /home/iq2lc/.d-rats-ev
INFO:MainApp:_refresh_lang: Setting language to: it
INFO:MainApp:Python pycountry package needed for selecting languages.
INFO:MainApp:D-RATS v0.4.1.dev0 starting at Tue Mar 25 09:43:39 2025 on Platform UnixPlatform:
  configuration:       /home/iq2lc/.d-rats-ev
  system_data: /home/iq2lc/Scaricati/D-Rats-master
  OS version: Linux iq2lclinux 6.1.42-antix.1-amd64-smp #1 SMP PREEMPT_DYNAMIC Mon Jul 31 18:30:05 EEST 2023 x86_64
INFO:MainApp:load position from config file
INFO:MainApp:_static_gps: Configuring the Static position: 45.837910,9.404950
INFO:MainApp:_static_gps: Mapserver not active: False, call: IQ2LC
INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8379,9.4049@0 ft at 09:43:39
INFO:D-Rats:main: reloading app

INFO:MainApp:_refresh_lang: Setting language to: it
INFO:MainApp:Python pycountry package needed for selecting languages.
INFO:MainApp:load the main window
INFO:MainApp:load main window with self config

(d-rats.py:32605): Gtk-WARNING **: 09:43:40.151: Attempting to add a widget with type GtkBox to a GtkMenuItem, but as a GtkBin subclass a GtkMenuItem can only contain one widget at a time; it already contains a widget of type GtkAccelLabel
INFO:DRatsPluginServer:Started serve_forever() thread
INFO:MessageRouter:_route_via_station: Starting message router thread
INFO:MainApp:connect main window
INFO:MainApp:connect tabs
INFO:MainApp:going online
INFO:MessageRouter:_run_one: Station list was empty
INFO:MainApp:create map window object-----
INFO:MapWindow:init MapWindow
INFO:MainApp:create map window object: connect object-----
INFO:MainApp:query local gps device to see our current position
INFO:MainApp:invoke config refresh
INFO:MainApp:Refreshing config...
INFO:gps:set_units: Set GPS units to Metric
INFO:MainApp:refresh comms
INFO:MainApp:_refresh_comms: Re-Starting ports_0
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_1
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_2
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_3
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_4
INFO:MainApp:Starting Comms
INFO:MainApp:start_comms: Starting port ports_4 (ttyUSB0)
INFO:SWFSerial:Software XON/XOFF control initialized
INFO:MainApp:_refresh_comms: Re-Starting ports_5
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_6
INFO:MainApp:Starting Comms
INFO:MainApp:check_comms_status: Ports expected to be already started:
INFO:MainApp:check_comms_status: ttyUSB0
INFO:MainApp:check_comms_status: Checking all Ports from config:
INFO:MainApp:check_comms_status: portid ports_0
INFO:MainApp:check_comms_status: Port Local_Ratflector not started
INFO:MainApp:check_comms_status: portid ports_1
INFO:MainApp:check_comms_status: Port RAT not started
INFO:MainApp:check_comms_status: portid ports_2
INFO:MainApp:check_comms_status: Port sttammany.ratflector.com not started
INFO:MainApp:check_comms_status: portid ports_3
INFO:MainApp:check_comms_status: Port ttyS0 not started
INFO:MainApp:check_comms_status: portid ports_4
INFO:MainApp:check_comms_status: Port ttyUSB0 already started!
INFO:MainApp:check_comms_status: portid ports_5
INFO:MainApp:check_comms_status: Port ttyUSB01 not started
INFO:MainApp:check_comms_status: portid ports_6
INFO:MainApp:check_comms_status: Port ttyUSB02 not started
INFO:MainApp:_refresh_gps : GPS: False on @4800
INFO:MainApp:_static_gps: Configuring the Static position: 45.837910,9.404950
INFO:MainApp:_static_gps: Mapserver not active: False, call: IQ2LC
INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8379,9.4049@0 m at 09:43:40
INFO:MainApp:_refresh_map: reconfigure Mapwindow with new map
INFO:MainApp:_refresh_map: reconfigured mapurl to: https://tile.openstreetmap.de/
INFO:MainApp:_refresh_map: reconfigured mapkey to: None
INFO:MainApp:_refresh_lang: Setting language to: it
INFO:MainApp:Python pycountry package needed for selecting languages.
INFO:StationsList:_menu_handler: executing ping all
INFO:StationsList:_menu_handler: Doing CQCQCQ ping on port ttyUSB0
INFO:Transporter:send_frames: [SERIAL /dev/ttyUSB0@9600] Waiting 1.7 sec before transmitting
INFO:StationsList:_menu_handler: executing ping all
INFO:StationsList:_menu_handler: Doing CQCQCQ ping on port ttyUSB0
INFO:Transporter:send_frames: [SERIAL /dev/ttyUSB0@9600] Waiting 1.3 sec before transmitting
INFO:Transporter:_parse_gps: [SERIAL /dev/ttyUSB0@9600] Found GPS string: bytearray(b'$GPGGA,000446.028,,,,,0,00,,,M,0.0,M,,0000*5A\r\n$GPRMC,000446.028,V,,,,,,,080810,,,N*40\r\n IZ2LXI ,                    ')
INFO:gps:parse_GPS: exception: could not convert string to float: ''
INFO:gps:parse_GPS: exception: could not convert string to float: ''
INFO:SessionManager:incoming: Received block 0:0 for session `chat'
INFO:SessionManager:incoming: Received block 0:0 for session `chat'

```

CHANGING THE MY STATION ON ID51 TO IQ2LC4
```

INFO:Transporter:_parse_gps: [SERIAL /dev/ttyUSB0@9600] Found GPS string: bytearray(b'$GPGGA,085440.428,,,,,0,00,,,M,0.0,M,,0000*55\r\n$GPRMC,085441.428,V,,,,,,,100810,,,N*47\r\nIQ2LC4  ,                    ')
INFO:gps:parse_GPS: exception: could not convert string to float: ''
INFO:gps:parse_GPS: exception: could not convert string to float: ''
INFO:SessionManager:incoming: Received block 0:0 for session `chat'
INFO:Transporter:_parse_gps: [SERIAL /dev/ttyUSB0@9600] Found GPS string: bytearray(b'$GPGGA,085442.428,,,,,0,00,,,M,0.0,M,,0000*57\r\n$GPRMC,085442.428,V,,,,,,,100810,,,N*44\r\nIQ2LC4  ,                    ')
INFO:gps:parse_GPS: exception: could not convert string to float: ''
INFO:gps:parse_GPS: exception: could not convert string to float: ''
INFO:SessionManager:incoming: Received block 0:0 for session `chat'
INFO:SessionManager:incoming: Received block 0:0 for session `chat'
^Cmainapp/ev_shutdown
mainapp/do_shutdown
INFO:MainApp:Saving config...
INFO:Transporter:send_frames: [SERIAL /dev/ttyUSB0@9600] Waiting 0.8 sec before transmitting
INFO:SWFSerial:is_xon: Aiee! Read a non-XOFF char: `b'\x13'`
INFO:SWFSerial:is_xon: Assuming IXANY behavior

```

maurizioandreotti, 2025-03-25:
opening Drats on antix with 2820 and leaving it in listening
putting ID51 with GPS fixed in auto TX, I see the call IQ2LC4 appear on the station list:

```
iq2lc@iq2lclinux:~/Scaricati/D-Rats-master
$ ./d-rats.py 

(d-rats.py:8884): dbind-WARNING **: 10:40:14.004: Couldn't connect to accessibility bus: Failed to connect to socket /run/user/0/at-spi/bus_0.0: Permesso negato
INFO:DratsConfig:File /home/iq2lc/.d-rats-ev/d-rats.config
INFO:QST:FeedParser not available
INFO:Spelling:Demand-opening aspell...
INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
INFO:D-Rats:main: args.config = /home/iq2lc/.d-rats-ev
INFO:MainApp:_refresh_lang: Setting language to: it
INFO:MainApp:Python pycountry package needed for selecting languages.
INFO:MainApp:D-RATS v0.4.1.dev0 starting at Tue Mar 25 10:40:14 2025 on Platform UnixPlatform:
  configuration:       /home/iq2lc/.d-rats-ev
  system_data: /home/iq2lc/Scaricati/D-Rats-master
  OS version: Linux iq2lclinux 6.1.42-antix.1-amd64-smp #1 SMP PREEMPT_DYNAMIC Mon Jul 31 18:30:05 EEST 2023 x86_64
INFO:MainApp:load position from config file
INFO:MainApp:_static_gps: Configuring the Static position: 45.837910,9.404950
INFO:MainApp:_static_gps: Mapserver not active: False, call: IQ2LC
INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8379,9.4049@0 ft at 10:40:14
INFO:D-Rats:main: reloading app

INFO:MainApp:_refresh_lang: Setting language to: it
INFO:MainApp:Python pycountry package needed for selecting languages.
INFO:MainApp:load the main window
INFO:MainApp:load main window with self config

(d-rats.py:8884): Gtk-WARNING **: 10:40:14.561: Attempting to add a widget with type GtkBox to a GtkMenuItem, but as a GtkBin subclass a GtkMenuItem can only contain one widget at a time; it already contains a widget of type GtkAccelLabel
INFO:DRatsPluginServer:Started serve_forever() thread
INFO:MessageRouter:_route_via_station: Starting message router thread
INFO:MainApp:connect main window
INFO:MainApp:connect tabs
INFO:MainApp:going online
INFO:MainApp:create map window object-----
INFO:MessageRouter:_run_one: Station list was empty
INFO:MapWindow:init MapWindow
INFO:MainApp:create map window object: connect object-----
INFO:MainApp:query local gps device to see our current position
INFO:MainApp:invoke config refresh
INFO:MainApp:Refreshing config...
INFO:gps:set_units: Set GPS units to Metric
INFO:MainApp:refresh comms
INFO:MainApp:_refresh_comms: Re-Starting ports_0
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_1
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_2
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_3
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_4
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_5
INFO:MainApp:Starting Comms
INFO:MainApp:start_comms: Starting port ports_5 (ttyUSB01)
INFO:MainApp:_refresh_comms: Re-Starting ports_6
INFO:MainApp:Starting Comms
INFO:MainApp:check_comms_status: Ports expected to be already started:
INFO:MainApp:check_comms_status: ttyUSB01
INFO:MainApp:check_comms_status: Checking all Ports from config:
INFO:MainApp:check_comms_status: portid ports_0
INFO:MainApp:check_comms_status: Port Local_Ratflector not started
INFO:MainApp:check_comms_status: portid ports_1
INFO:MainApp:check_comms_status: Port RAT not started
INFO:MainApp:check_comms_status: portid ports_2
INFO:MainApp:check_comms_status: Port sttammany.ratflector.com not started
INFO:MainApp:check_comms_status: portid ports_3
[... 57 lines cut ...]
INFO:SessionManager:incoming: Received block 0:0 for session `chat'
INFO:MainApp:__incoming_gps_fix: aprs_code missing - forced to: \? 
INFO:MainApp:__incoming_gps_fix: Export to external mapserver not active: False
INFO:StaticGPSSource:StaticGPSPosition: IQ2LC4 reporting 45.8004,9.4283@216 m at 10:44:19
INFO:Transporter:worker: [SERIAL /dev/ttyUSB1@9600] ### Unconverted data: b'\r\n'
INFO:Transporter:_parse_gps: [SERIAL /dev/ttyUSB1@9600] Found GPS string: bytearray(b'$GPGGA,094528.000,4548.0226,N,00925.6976,E,1,06,1.2,215.2,M,48.0,M,,0000*58\r\n$GPRMC,094529.000,A,4548.0226,N,00925.6976,E,0.00,103.70,250325,,,A*60\r\nIQ2LC4  ,                    ')
INFO:NMEAGPSPosition:_parse_gpgga:  45.800377,9.428293
INFO:NMEAGPSPosition:_parse_gprmc: GPRMC is valid
INFO:gps:parse_GPS: Appending fix: IQ2LC4 reporting 45.8004,9.4283@0 m at 09:45:29
INFO:SessionManager:incoming: Received block 0:0 for session `chat'
INFO:MainApp:__incoming_gps_fix: aprs_code missing - forced to: \? 
INFO:MainApp:__incoming_gps_fix: Export to external mapserver not active: False
INFO:StaticGPSSource:StaticGPSPosition: IQ2LC4 reporting 45.8004,9.4283@215 m at 10:45:19
INFO:Transporter:worker: [SERIAL /dev/ttyUSB1@9600] ### Unconverted data: b'\r\n'
INFO:Transporter:_parse_gps: [SERIAL /dev/ttyUSB1@9600] Found GPS string: bytearray(b'$GPGGA,094628.000,4548.0225,N,00925.6976,E,1,06,1.2,215.2,M,48.0,M,,0000*58\r\n$GPRMC,094629.000,A,4548.0225,N,00925.6976,E,0.00,103.70,250325,,,A*60\r\nIQ2LC4  ,                    ')
INFO:NMEAGPSPosition:_parse_gpgga:  45.800375,9.428293
INFO:NMEAGPSPosition:_parse_gprmc: GPRMC is valid
INFO:gps:parse_GPS: Appending fix: IQ2LC4 reporting 45.8004,9.4283@0 m at 09:46:29
INFO:SessionManager:incoming: Received block 0:0 for session `chat'
INFO:MainApp:__incoming_gps_fix: aprs_code missing - forced to: \? 
INFO:MainApp:__incoming_gps_fix: Export to external mapserver not active: False
INFO:StaticGPSSource:StaticGPSPosition: IQ2LC4 reporting 45.8004,9.4283@215 m at 10:46:19
INFO:Transporter:worker: [SERIAL /dev/ttyUSB1@9600] ### Unconverted data: b'\r\n'

```

maurizioandreotti, 2025-03-25:
done a test from mobaxterm on windows 10 from different PC

  25/03/2025   13:28.25   /home/mobaxterm  ./d-rats
/usr/local/lib/pypi/pydub/utils.py:170: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
  warn("Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work", RuntimeWarning)
INFO:Spelling:Demand-opening aspell...
INFO:Spelling:Tested spelling okay: ['spelling,', 'spelino,', 'peeling,', 'soling,', 'spring,', 'sterling,', 'spline,', 'feeling']
INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
INFO:D-Rats:main: args.config = /home/mobaxterm/.d-rats-ev
INFO:DratsConfig:File /home/mobaxterm/.d-rats-ev/d-rats.config
INFO:MainApp:_refresh_lang: Setting language to: English
INFO:MainApp:set_locale: OS Locale set to: en_US.UTF-8
INFO:MainApp:D-RATS v0.4.1.dev24-ga6b7e1f-dirty starting at Tue Mar 25 13:28:52 2025 on Platform UnixPlatform:
  configuration:       /home/mobaxterm/.d-rats-ev
  system_data: /drives/c/Users/mauri/AppData/Roaming/MobaXterm/home/d-rats-git
  OS version: CYGWIN_NT-10.0-19045-WOW64 Mao-IdeaPad-500 3.3.5-341.i686 2022-06-12 08:16 UTC i686
INFO:MainApp:load position from config file
INFO:MainApp:_static_gps: Configuring the Static position: 45.801730,9.426220
INFO:MainApp:_static_gps: Mapserver not active: False, call: IZ2LXI-2
INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8017,9.4262@0 ft at 13:28:52
INFO:D-Rats:main: reloading app

INFO:MainApp:_refresh_lang: Setting language to: English
INFO:MainApp:set_locale: OS Locale set to: en_US.UTF-8
INFO:MainApp:load the main window

(d-rats.py:927): GLib-GIO-CRITICAL **: g_dbus_proxy_new_sync: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
INFO:MainApp:load main window with self config
INFO:LocalFileV:refresh: Adding local file `/home/mobaxterm/D-RATS Shared/3_Mode Overview.pdf'
INFO:LocalFileV:refresh: Adding local file `/home/mobaxterm/D-RATS Shared/Hello.txt'

(d-rats.py:927): Gtk-WARNING **: Attempting to add a widget with type GtkBox to a GtkMenuItem, but as a GtkBin subclass a GtkMenuItem can only contain one widget at a time; it already contains a widget of type GtkAccelLabel
INFO:DRatsPluginServer:Started serve_forever() thread
INFO:MessageRouter:_route_via_station: Starting message router thread
INFO:MainApp:connect main window
INFO:MainApp:connect tabs
INFO:MainApp:going online
INFO:MainApp:create map window object-----
INFO:MapWindow:init MapWindow
INFO:MessageRouter:_run_one: Station list was empty
INFO:MainApp:create map window object: connect object-----
INFO:MainApp:query local gps device to see our current position
INFO:MainApp:invoke config refresh
INFO:MainApp:Refreshing config...
INFO:gps:set_units: Set GPS units to Metric
INFO:MainApp:refresh comms
INFO:MainApp:_refresh_comms: Re-Starting ports_0
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_1
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_2
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_3
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_4
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_5
INFO:MainApp:Starting Comms
INFO:MainApp:_refresh_comms: Re-Starting ports_6
INFO:MainApp:Starting Comms
INFO:MainApp:start_comms: Starting port ports_6 (ttyS5)
INFO:MainApp:check_comms_status: Ports expected to be already started:
INFO:SWFSerial:Software XON/XOFF control initialized
INFO:MainApp:check_comms_status: ttyS5
INFO:MainApp:check_comms_status: Checking all Ports from config:
INFO:MainApp:check_comms_status: portid ports_0
INFO:MainApp:check_comms_status: Port alabama.ratflector.com not started
INFO:MainApp:check_comms_status: portid ports_1
INFO:MainApp:check_comms_status: Port k3pdr.dstargateway.org not started
INFO:MainApp:check_comms_status: portid ports_2
INFO:MainApp:check_comms_status: Port pldares.ratflector.com not started
INFO:MainApp:check_comms_status: portid ports_3
INFO:MainApp:check_comms_status: Port RAT not started
INFO:MainApp:check_comms_status: portid ports_4
INFO:MainApp:check_comms_status: Port sewx.ratflector.com not started
INFO:MainApp:check_comms_status: portid ports_5
INFO:MainApp:check_comms_status: Port sttammany.ratflector.com not started
INFO:MainApp:check_comms_status: portid ports_6
INFO:MainApp:check_comms_status: Port ttyS5 already started!
INFO:MainApp:_refresh_gps : GPS: False on @4800
INFO:MainApp:_static_gps: Configuring the Static position: 45.801730,9.426220
INFO:MainApp:_static_gps: Mapserver not active: False, call: IZ2LXI-2
INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8017,9.4262@0 m at 13:28:54
INFO:MainApp:_refresh_map: reconfigure Mapwindow with new map
INFO:MainApp:_refresh_map: reconfigured mapurl to: https://tile.thunderforest.com/landscape/
INFO:MainApp:_refresh_map: reconfigured mapkey to: ?apikey=5a1a4a79354244a38707d83969fd88a1
INFO:MainApp:_refresh_lang: Setting language to: English
INFO:MainApp:set_locale: OS Locale set to: en_US.UTF-8
INFO:StationsList:_menu_handler: executing ping all
INFO:StationsList:_menu_handler: Doing CQCQCQ ping on port ttyS5
INFO:Transporter:send_frames: [SERIAL /dev/ttyS5@9600] Waiting 2.0 sec before transmitting
INFO:Transporter:_parse_gps: [SERIAL /dev/ttyS5@9600] Found GPS string: bytearray(b'$GPGGA,122914.473,,,,,0,00,,,M,0.0,M,,0000*5B\r\n$GPRMC,122914.473,V,,,,,,,100810,,,N*48\r\nIQ2LC4  ,                    ')
INFO:gps:parse_GPS: exception: could not convert string to float: ''
INFO:gps:parse_GPS: exception: could not convert string to float: ''
INFO:SessionManager:incoming: Received block 0:0 for session `chat'
INFO:Transporter:worker: [SERIAL /dev/ttyS5@9600] ### Unconverted data: b'\r\n'

wb8tyw, 2025-03-25:
The unit test section of gps.py needs to be modified so that it can do test parses of the GPS strings extracted from the log.
There is probably a problem with the regex based on the error messages about converting string to float.   That should not be too hard to fix.

The message about the unconverted data is curious as to why it is being written.

It looks like it may be trying to parse 3 lines where it may be expecting only one.

wb8tyw, 2025-03-26:
How the GPS module should be working:
[RedHat Architects Guide GPS and GPS data formats](https://www.redhat.com/en/blog/architects-guide-gps-and-gps-data-formats)

There are several GPS python packages, not sure which is the best.  Most of them seem to only be available on PyPi

RedHat based OSes has a python3-gpsd packages that communicates between a GPS device and makes it available for a client.   Unknown yet if that package contains a client.

Debian based OSes have a python3-gps package which is described as a client to a GPSD server.   Unknown if this contains a client.

wb8tyw, 2025-03-27:
Looks like some of the fields that the D-Rats GPS parser expects to find are not aways present.

From the translated D-Star documentation, the GPS string should have these parts.  I will need to check to see how this matches the code.

`1.  NMEA  sentence:  You  can  send  one  to  three  NMEA  sentences.
 2.  Callsign  (fixed  to  8  characters,  unused  area  filled  with  spaces)  
3.  Message  (fixed  to  20  characters,  unused  area  filled  with  spaces`

wb8tyw, 2025-03-28:
Adding these strings to the gps.py test module has reproduced the failure.

Transport has a different GPS parser than the gps.py module.

wb8tyw, 2025-03-28:
It looks like gps.py is not able to handle a GPS string from a real radio.

This code is going to take some time to untangle and make compliant with the the standards that I have recently located.
It is also apparently duplicating a lot of code that is in existing Python libraries.

So is this something that can be moved to be fixed in the drats2 project.