ham-radio-software/D-Rats #307: Issue occurs when user clicks on 'preferences'

beenej, 2025-04-15:
See image below. I am sharing this on behalf of my father. He is a die hard HAM Radio with his General license. This is the debug log from the error that he can recreate when clicking on preferences. If he chooses to "ignore all", preferences menu is null (empty screen).

[image]

For more information, feel free to leave contact info for him or I to reach out on here and we will do so with more info. Thanks!

wb8tyw, 2025-04-16:
I am having trouble finding time to properly chase this down.

I would recommend that you switch to [D-RATS on Microsoft Windows with MobaXterm](https://github.com/ham-radio-software/D-Rats/wiki/010.020-Installation-of-D%E2%80%90Rats-on-Microsoft-Windows-with-MobaXterm) which seems to be simpler to setup than msys2.

* MSYS2 currently requires Windows 11 and Windows 10, but will probably drop Windows 10 this year.

This failure message appears to  a secondary failure from the primary failure.

The secondary error message is indicating that the MSYS2 Linux emulation environment is missing a GUI text editor, and I am still trying to find time to debug that issue.

The primary error message is indicating that there is invalid information in the drats config file.   D-Rats uses an older and vary common config file format, but then it expects that the data in some of the fields be able to be parsed, but does not always enforce that when you enter the data into D-Rats that the data entered is valid.

The indication is that here is an invalid value in the list of "radios" which can be fixed by editing that file with a text editor, or deleting the file and going through the configuration again.    The usual problem is things like punctuation characters in the radio description.
The radio description which needs to be in the format below.  Note that the number of commas are significant.

```text
[ports]
ports_0 = True,net:pldares.ratflector.com:9000,,True,True,pldares
ports_1 = True,/dev/ttyUSB0,9600,False,False,Icom Radio
```

beenej, 2025-04-17:
I will come back to this as it's late, but thank you for your answer! I'll pass this on. Maybe MOBA works better for him.